daid/LegacyCura #1069: Cura crash if layer view is changed before loading a first model

pmsimard, 2015-01-07:
It seems Cura will crash if you do the following steps:
- Open Cura
- Go in Layer view
- Click on the layer scroll bar (value will go to layer 0)
- Load a model (while still in layer view)
- Crash

Even if I was attached to the process, python would not break to the place of the crash. I do not know where this is crashing.

pmsimard, 2015-01-09:
I'm testing pycharm and it's better then what I had with Python for Visual Studio. It works well except for 2-3 things:
- I managed to configure the Run properly to trigger Cura directly but it wont attach to the process automatically so I cannot hit the breakpoints I would set during the boot. I can attach to the process afterward and its ok. Do you have any luck on that side?
- When pycharm hit a break point cura will stall like it should but I'll get no clues (change of focus, blinking window, ...) that pycharm is on a breakpoint.
- Is there a Visual Studio control scheme ? :)

daid, 2015-01-12:
My configuration is like this:
http://i.imgur.com/VJM2B12.png
* Nothing special, and I can set breakpoints. But when I run as debug I can set breakpoints.
* Yes, that is a bit of a "oh, must be a breakpoint" moment. Note that the console logging in pycharm is also very valuable. Most of the time I do not use the debugger at all, and just the console log.