daid/LegacyCura #1073: Specify the license

traverseda, 2015-01-07:
In theory it's AGPL, as that's what cura engine is licensed under and it depends on cura engine. But it would be nice to have the licensing specified.

daid, 2015-01-07:
Help -> About specifies all licenses used.
All files contain copyright notices: Copyright (C) 2013 David Braam - Released under terms of the AGPLv3 License
License file is distributed with the source and the release.

What else do you want?

closed 2015-01-07 by daid

traverseda, 2015-01-07:
A license.txt is traditional.

daid, 2015-01-07:
https://github.com/daid/Cura/blob/SteamEngine/Cura/LICENSE
https://github.com/Ultimaker/CuraEngine/blob/master/LICENSE
https://github.com/Ultimaker/Marlin/blob/master/Marlin/COPYING
https://github.com/Ultimaker/Ultimaker2Marlin/blob/master/Marlin/COPYING