daid/LegacyCura #1075: Strange shadow patterns for largish model

hallzhallz, 2015-01-07:
When I load this model ( http://www.thingiverse.com/thing:39675/#files ) there is a large triangle shadow shooting off the bed. If I load a simple model and then this model also at the same time then the shadow disappears completely (but this might just be to do with the multi-object printing?).

When slicing the object to gcode it goes fine up until about halfway up the model and then the print head loses alignment and prints spaghetti on top of the print.

Skirt and support are disable so I am not sure what is going on, their should not be any shadows so to speak.

[image]
[image]

[image]

daid, 2015-01-08:
At which size did you scale this model?

hallzhallz, 2015-01-10:
Hi Daid,

I did not scale or rotate the model in any way. It is just how it appears when you first load it. I am using windows 7 with v14.12.1 but you have probably determined that from the screenshots.

It does not effect the printing in any way so maybe it is something to do with my system/drivers.

I'll try updating them my drivers and report back.

Cheers, David.

daid, 2015-01-11:
Could be a driver thing, or a combined Cura bug and driver thing. As I'm not seeing this bug on my laptop.

hallzhallz, 2015-01-14:
I tried updated drivers but it did not resolve the problem.

I am not sure that I said in the original descirption that the shadow appears when you click/select the object.

Given that it just seems to be on my machine and doesn't affect printing I will close the issue.

Thanks for taking a look. Cheers.

closed 2015-01-14 by hallzhallz