daid/LegacyCura #1078: Gcode preview shifted glitch

scottlawsonbc, 2015-01-09:
This issue only appeared today, and I have never encountered it before. The gcode preview is glitched and shifted horizontally. You can see this in the image below:

[image]

This issue persists:
* When various settings like layer height, infill, etc are changed
* For all .stl files that I have tried, even ones that have previously not had this issue

However, when I save the gcode file and view it in Repetier, it looks like Cura is actually slicing the model correctly.

[image]

Settings used:
````
[profile]
layer_height = 0.2
wall_thickness = 1.6
retraction_enable = True
solid_layer_thickness = 1.4
fill_density = 10
nozzle_size = 0.4
print_speed = 50
print_temperature = 230
print_temperature2 = 0
print_temperature3 = 0
print_temperature4 = 0
print_bed_temperature = 60
support = None
platform_adhesion = None
support_dual_extrusion = Both
wipe_tower = False
wipe_tower_volume = 15
ooze_shield = False
filament_diameter = 1.75
filament_diameter2 = 0
filament_diameter3 = 0
filament_diameter4 = 0
filament_flow = 94
retraction_speed = 40
retraction_amount = 3
retraction_dual_amount = 16.5
retraction_min_travel = 1.5
retraction_combing = True
retraction_minimal_extrusion = 0.02
retraction_hop = 0.1
bottom_thickness = 0.3
layer0_width_factor = 100
object_sink = 0
overlap_dual = 0.15
travel_speed = 200
bottom_layer_speed = 15
infill_speed = 0.0
inset0_speed = 30
insetx_speed = 0
cool_min_layer_time = 5
fan_enabled = True
skirt_line_count = 1
skirt_gap = 4
skirt_minimal_length = 100
fan_full_height = 0.4
fan_speed = 30
fan_speed_max = 30
cool_min_feedrate = 10
cool_head_lift = False
solid_top = True
solid_bottom = True
fill_overlap = 15
support_type = Lines
support_angle = 27
support_fill_rate = 15
support_xy_distance = 1
support_z_distance = 0.15
spiralize = False
simple_mode = False
brim_line_count = 20
raft_margin = 5.0
raft_line_spacing = 4
raft_base_thickness = 0.2
raft_base_linewidth = 2
raft_interface_thickness = 0.3
raft_interface_linewidth = 0.6
raft_airgap = 0.22
raft_surface_layers = 2
fix_horrible_union_all_type_a = False
fix_horrible_union_all_type_b = False
fix_horrible_use_open_bits = False
fix_horrible_extensive_stitching = False
plugin_config = (lp1
	(dp2
	S'params'
	p3
	(dp4
	S'targetL'
	p5
	V1
	sS'extruderTwo'
	p6
	V213
	p7
	sS'flowrateTwo'
	p8
	V
	sS'targetZ'
	p9
	V
	sS'flowrate'
	p10
	V
	sS'fanSpeed'
	p11
	V150
	p12
	sS'platformTemp'
	p13
	V
	sS'speed'
	p14
	V
	sS'flowrateOne'
	p15
	V
	sS'extruderOne'
	p16
	V2
	ssS'filename'
	p17
	S'TweakAtZ.py'
	p18
	sa.
object_center_x = -1
object_center_y = -1
````

daid, 2015-01-09:
Which version of Cura?

scottlawsonbc, 2015-01-09:
My apologies, I should have included that information. This is with Cura 14.09. Let me know if there is any other useful data that I could provide.

daid, 2015-01-09:
Can you try the latest version? As a problem in the GCode viewer caused this if you used G92 to offset the build area in the start code.

scottlawsonbc, 2015-01-09:
I will give it a try it on the latest development version, I believe I'm currently using the latest release version. I also don't know why this is suddenly happening since it didn't happen ever before and I can't think of anything that has changed.

Start gcode:

````
M104 S{print_temperature}
M140 S{print_bed_temperature}

; Home axes
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28 Z0     ;move Z to min endstops

; Move closer to starting position
G1 Z15.0 F{travel_speed} ;move the platform down 15mm

; Wait for temperatures to be reached
M190 S{print_bed_temperature} 
M109 S{print_temperature} 
G92 E0              
G1 F200 E3       
G92 E0         
G1 F{travel_speed}

;Put printing message on LCD screen
M117 Lawson printing...
````

End gcode:
```
;End GCode
G91                                    ;relative positioning
G1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure
G1 Z+0.5 E-5 X-20 Y-20 F{travel_speed} ;move Z up a bit and retract filament even more
G28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way
G28 Z
; Only turn off heaters after homing in case you want to print again
M104 S0                     ;extruder heater off
M140 S0                     ;heated bed heater off (if you have it)
M84                         ;steppers off
G90                         ;absolute positioning
;{profile_string}
```
Machine settings:
[image]

daid, 2015-01-09:
There is a 14.12.1 release. (But the update notification has not been activated due to the new UM2 firmware giving some people issues)

There are also beta versions at:
http://software.ultimaker.com/Cura_closed_beta/

scottlawsonbc, 2015-01-10:
Thanks for the information about 14.12.1, what a great update! 

I've managed to narrow the issue down a bit more. It seems to be caused by something in my custom settings profile but I'm not sure exactly what line/setting is causing the problem. I've posted the full text of my [settings.ini file here](http://pastebin.com/DUf6Vjzf). There are several start/end gcodes in that file, but I am currently only using `start.gcode` and `end.gcode`.

Interestingly, the same profile does not cause the issue in an older version of Cura. 

Using Cura 14.01:
**Default profile:** Cannot replicate issue
**Custom profile:** Cannot replicate issue

Using Cura 14.09:
**Default profile:** Cannot replicate issue
**Custom profile:** Issue encountered

Using Cura 14.12.1:
**Default profile:** Cannot replicate issue
**Custom profile:** Issue encountered

To narrow it down further, I found a copy of my settings profile from a few weeks ago. This profile does **not** cause the issue on any version of Cura. I've posted the full text here of this older [settings.ini file here](http://pastebin.com/7L7MmHys).

You can see a [diff of the two settings.ini files here](https://www.diffchecker.com/36ttx8qg).

Are you able to replicate the issue using my `settings.ini` file?

daid, 2015-01-10:
Found the problem.

Your tweakAtZ configuration sets a 2nd nozzle temperature, causing Cura to incorrectly think you are using the second nozzle from that point on and causing an offset in the shown GCode. The lower part is shown correctly as that's drawn with information directly from the engine instead of the gcode.

commit 2f728760 (2015-01-10): Fixed #1078
closed 2015-01-10 by daid