daid/LegacyCura #1080: 15.01RC7: Multiple objects

Dim3nsioneer, 2015-01-11:
When I load multiple objects into Cura (tested with a 1cm cube which I multiplied once), all objects are sliced but only one is saved into gcode.

daid, 2015-01-11:
With or without plugins?

Dim3nsioneer, 2015-01-11:
Both...

daid, 2015-01-12:
I cannot seem to reproduce this, it always slices correctly for me, even if I load files and switch modes.

What OS are you running on again? (OSX is giving me strange issues from time to time)

Dim3nsioneer, 2015-01-12:
Win 8.0 and win8.1
You want me to slice a specific situation?

Von Samsung Mobile gesendet

daid, 2015-01-12:
Are your machine settings default?

When I load my "20x20x10 cube" twice, I get 2 objects in the GCode.

Dim3nsioneer, 2015-01-12:
Yes, I quickly setup a new UM2 machine. Doesn't work.
And I realised something else. Cura doesn't see the SDCard when plugged in (and Windows immediately opened an Explorer window). I had that one once in 2013 I think. Then it was possible to get rid of it by reinstalling Cura. That trick didn't work this time. All Cura versions are affected (back to something like 14.01). Any idea what might cause such an issue? Something with Python?

Dim3nsioneer, 2015-01-12:
And I went back in Windows System Restore as far back as I still have the data (I cleared it recently unfortunately). No effect.

colorstone, 2015-01-15:
look this 
[image]

colorstone, 2015-01-15:
what's all this ??

daid, 2015-01-15:
A finger, a print, a few blobs, and some shadows.

Dim3nsioneer, 2015-01-15:
Couldn't have written it more precise. Though, still don't see the connection to the topic of this issue. Maybe a misplacing by @colorstone ?

colorstone, 2015-01-16:
This problem occurs in the 2th Shell Move to StartPoint (out shell StartPoint)
a little pla point has be Overflow

colorstone, 2015-01-16:
fill density 30%  print speed 30  temperature 200

colorstone, 2015-01-16:
a testmodel 
[image]
[image]

Dim3nsioneer, 2015-01-16:
@colorstone : As you don't seem to understand Daid's cynism and my message: please open a new issue if you have something to report and don't put your stuff into someone else's report which is about something completely else. Thank you.

colorstone, 2015-01-16:
oh~ sorry, I don't know...
how to open a new issue? @Dim3nsioneer 
i can't click the submit new issue button

Dim3nsioneer, 2015-01-16:
https://github.com/daid/Cura/issues/new

Dim3nsioneer, 2015-01-19:
@daid : I had a closer look at the issue. It looks as if the gcode is generated and also displayed in the layer view of Cura as 'one-at-a-time' (as selected) but saved as 'all-at-once'.
Does that make any sense to you?

daid, 2015-01-19:
That would make no sense at all... as what is shown in the layer view comes out of the engine, while it's generating the layers...

Dim3nsioneer, 2015-01-20:
Mistery solved. It was my fault. I didn't realize that the layer numbering starts from 0 for each object. I must have missed this change as some time ago Cura started from 0 and increased the layer number throughout the different objects.

closed 2015-01-20 by Dim3nsioneer