daid/LegacyCura #1081: Print one at a time not working only prints all at ones

RENZIN, 2015-01-12:
I'm trying to print one at a time but it doesnt work Only prints all at ones.  Can anybody help me?

Dim3nsioneer, 2015-01-12:
Clear the platform, switch to print-one-at-a-time and reload the objects. It seems the setting is a bit buggy. But @daid will anyway have a look at the setting due to some other issue ( #1080 ).

daid, 2015-01-12:
What kind of printer are you using? You might need to setup the head size and gantry height in the machine settings. They are required for "one at a time" mode, and are only set by default on the Ultimakers.

closed 2015-01-12 by daid

RENZIN, 2015-01-12:
I'm using the Ultimaker 2 with default settings.  What are the settings I need for the ultimaker to make the print one at a time function work?

daid, 2015-01-12:
The settings are filled in by default for the Ultimaker2.

But the objects (all of them) need to be lower then 50mm, or else Cura will print all-at-once, because the objects will hit the gantry otherwise.

RENZIN, 2015-01-12:
GOOD!!!  I finally did lower mm and now is working.   Thank you