daid/LegacyCura #1091: Improper M106 numbers.

DarkAlchy, 2015-01-16:
Cura 14.12.1 and if I tell cura max is 100% and min is 50% I get M106 63 and M106 127 but if I ignore the red boxes when I enter 100% for min and 200% for max I get the real numbers in the Gcode of 127 and 255.

daid, 2015-01-16:
The fan speed is also gradually increased on the first few layers, so that might be the problem you're actually looking at.

DarkAlchy, 2015-01-17:
Negative there were only 3 M106 commands in the entire gcode I set up.  Layer 0 M106 S0, Layer 1 M106 S63 and Layer 2 M106 S127 (I set it up to have Layer 2+ full force).  As you can see those numbers reflect exactly 50% so I have to fake it with the red input box of 100% min and 200% max.

BagelOrb, 2015-04-14:
Cannot recreate issue. I'm seeing an M106 S255.

closed 2015-04-14 by BagelOrb

zhivko, 2015-12-29:
I would like to have fan turned on at 0 level - is it possible to set cura to do that ?
If I dont cool - the filament softens and stil while on 0 layer it becomes so soft it is expelled out of the filament path before it reaches the heater section...
So is there a way to turn it on already at layer 0 ?

BagelOrb, 2016-01-04:
What if you set the Fan speed min to 100% ? (Expert setting)

zhivko, 2016-01-04:
I found mistake - it was hardware nature - it's better I don't mention it here publicly ;)