daid/LegacyCura #1092: Model has a few blobs

colorstone, 2015-01-16:
This problem occurs in the 2th Shell Move to StartPoint (out shell StartPoint)
a little pla point has be Overflow
fill density 30% print speed 30 temperature 200

[image]
[image]
[image]

daid, 2015-01-17:
Which type of printer?

colorstone, 2015-01-17:
delta  ZXY  All

colorstone, 2015-01-18:
you can print it to check it   fill density 30% print speed 30 temperature 200

colorstone, 2015-01-18:
I'm trying to use makerbot printer to test

BagelOrb, 2015-04-14:
Please test this issue with a reliable printer. Try to test it with an Ultimaker! :D

closed 2015-04-14 by BagelOrb