daid/LegacyCura #1098: Cura 15.01-RC9 segfaults when trying to install custom firmware

SuperDoxin, 2015-01-20:
when using the machine -> install custom firmware tool a connecting dialog briefly shows before cura crashes. output on the terminal is as follows:

```
~$ cura
load preferences from /home/lieuwe/.cura/15.01-RC9/preferences.ini
Listening for engine communications on 49674
The program 'python' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadDrawable (invalid Pixmap or Window parameter)'.
  (Details: serial 60180 error_code 9 request_code 66 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
The program 'python' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadDrawable (invalid Pixmap or Window parameter)'.
  (Details: serial 60182 error_code 9 request_code 66 minor_code 0)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)
Segmentation fault
```

daid, 2015-01-20:
https://github.com/daid/Cura/blob/SteamEngine/Cura/gui/mainWindow.py#L573

Known issue, do not know what causes it. But only happens on certain linux installs.

SuperDoxin, 2015-01-20:
hm, the bug seems intermittent too. I tried running cura under PDB today but no sign of any problems. then I tried without pdb and again the firmware uploaded without issues.

might be a race condition in the threading somewhere, but with no way to reliably check it's impossible to know for sure.