daid/LegacyCura #1100: homebrew setup script fails to setup environment

phonyphonecall, 2015-01-20:
When setting up a new dev environment the homebrew script fails to execute properly.
Error:
```
$ brew install --build-bottle --fresh Cura/scripts/darwin/python.rb --universal
==> Downloading http://www.python.org/ftp/python/2.7.3/Python-2.7.3.tar.bz2
Already downloaded: /Library/Caches/Homebrew/python-2.7.3.tar.bz2
==> Downloading https://gist.github.com/raw/4349132/25662c6b382315b5db67bf949773d76471bbcee7/python-nfs-shutil.diff
######################################################################## 100.0%
==> Patching
patching file Lib/shutil.py
patching file Lib/test/test_shutil.py
patching file Modules/errnomodule.c
==> ./configure --prefix=/usr/local/Cellar/python/2.7.3 --enable-ipv6 --datarootdir=/usr/local/Cellar/python/2.7.3/share --datadir=/usr/local/Cellar/python/2.7.3/share --enable-framework=/usr/local/Ce
==> make
==> make install PYTHONAPPSDIR=/usr/local/Cellar/python/2.7.3
==> make frameworkinstallextras PYTHONAPPSDIR=/usr/local/Cellar/python/2.7.3/share/python
Error: wrong number of arguments (0 for 3)
Please report this bug:
    http://git.io/brew-troubleshooting
/usr/local/Library/Homebrew/formula.rb:86:in `initialize'
/Users/Scott/dev/github/Cura/scripts/darwin/python.rb:158:in `new'
/Users/Scott/dev/github/Cura/scripts/darwin/python.rb:158:in `install'
/usr/local/Library/Homebrew/build.rb:136:in `block in install'
/usr/local/Library/Homebrew/formula.rb:390:in `block in brew'
/usr/local/Library/Homebrew/formula.rb:745:in `block in stage'
/usr/local/Library/Homebrew/resource.rb:91:in `block in unpack'
/usr/local/Library/Homebrew/extend/fileutils.rb:21:in `mktemp'
/usr/local/Library/Homebrew/resource.rb:88:in `unpack'
/usr/local/Library/Homebrew/resource.rb:81:in `stage'
/usr/local/Library/Homebrew/formula.rb:743:in `stage'
/usr/local/Library/Homebrew/formula.rb:386:in `brew'
/usr/local/Library/Homebrew/build.rb:114:in `install'
/usr/local/Library/Homebrew/build.rb:185:in `<main>'
```
I've fixed all issues in brew doctor, and uninstalled everything python I could find on the machine. I would report it to the homebrew issues, but similar issues in that issue tracker claim that this error is a formulaic issue. Reference: https://github.com/Homebrew/homebrew/issues/33903

Let me know how I can help.

brew --verson: `0.9.5`
OSX `10.10.1`

daid, 2015-01-20:
Honestly, no clue. I'm glad my MacOS environenmt works. A few people have been trying to replicate it, but so far no-one has been successful. I have very little knowledge on this area, so there is little I can do to help here.

phonyphonecall, 2015-01-20:
@daid understandable. Do you have a suggested alternate method to setup a mac dev environment?

melka, 2015-01-22:
Here are the steps I followed to setup a build env on MacOS X 10.10.1, feel free to adapt it to your own setup.

Python
--------
```bash
brew uninstall python
brew install python --universal --framework
```

This should give you the necessary requirements for the non-system framework based universal python.

Virtualenv
--------

http://docs.python-guide.org/en/latest/dev/virtualenvs/

```bash
pip install virtualenv
virtualenv Cura
cd Cura
source bin/activate
```

Git
--------

http://stackoverflow.com/questions/791959/download-a-specific-tag-with-git
http://stackoverflow.com/questions/2411031/how-do-i-clone-into-a-non-empty-directory

```bash
cd Cura
git init
git remote add origin https://github.com/daid/Cura
git fetch
git checkout
````

wxPython
--------

http://umforum.ultimaker.com/index.php?/topic/7577-cura-mac-os-x-sources-compil/

download wxPython sources versions 3.0.1.1 [here](http://sourceforge.net/projects/wxpython/files/wxPython/3.0.1.1/wxPython-src-3.0.1.1.tar.bz2/download) and uncompress it somewhere. I did it in the Cura folder, and added the wxPython folder to .gitignore. You might get some build errors if you're on Yosemite, so apply the necessary changes from this link
https://github.com/KiCad/KicadOSXBuilder/issues/40

```bash
cd wxPython-src-3.0.1.1/wxPython
python build-wxpython.py --osx_cocoa --mac_arch=i386,x86_64 --install

Create file *Cura/bin/pythonw* with the following content:
```bash
#!/bin/bash
ENV=`python -c "import sys; print sys.prefix"`
PYTHON=`python -c "import sys; print sys.real_prefix"`/bin/python
export PYTHONHOME=$ENV
exec $PYTHON "$@"
```

Quit terminal, reopen, go back to Cura folder and reactivate virtualenv.

```bash
cd Cura-15.01-RC5-Makerbot
source bin/activate
```

power
--------

Download power 1.3 sources [here](https://github.com/Kentzo/Power/tree/v1.3
).

"super has no attribute init" build error
https://bitbucket.org/ronaldoussoren/pyobjc/issue/110/subclass-of-nsobject-super-has-no

```bash
cd power-1.3
python setup.py build
python setup.py install
```

Python packages
--------

```bash
cd Cura
pip install -r requirements.txt
pip install -r requirements_darwin.txt
```

modulegraph error
https://bitbucket.org/jinnko/py2app/commits/dba244c843397345d2bac8cdfc92ba9d66192bfc

Run in debug mode
--------

```bash
cd Cura
pythonw -m Cura.cura
```

Package Cura
--------

```bash
cd Cura
./package.sh darwin
```

phonyphonecall, 2015-01-22:
@melka thanks! I'll have a run through this.

3dht, 2015-08-10:
hello @melka and @daid  , thanks for your information . However do you guys have a problem of avr-gcc when you're building Cura on Mac ?? I just have installed avr-gcc4.9.2 with using brew but when executing ./package.sh darwin, I have an error:

CXX   MarlinSerial.cpp
In file included from SdFile.h:27:0,
                 from cardreader.h:8,
                 from Marlin_main.cpp:44:
SdBaseFile.h:38:8: error: using typedef-name 'fpos_t' after 'struct'
 struct fpos_t {
        ^
In file included from Marlin.h:10:0,
                 from Marlin_main.cpp:30:
/usr/local/Cellar/avr-gcc/4.9.2/avr/include/stdio.h:947:19: note: 'fpos_t' has a previous declaration here
 typedef long long fpos_t;
                   ^
make: *** [_UltimakerMarlin_250000/Marlin_main.o] Error 1
make: *** Waiting for unfinished jobs....

Wish you guys could help me , thank you in advance