daid/LegacyCura #1102: Unwanted Z and Y Jump with Cura 14.12.1 at 0.3mm

fcartegnie, 2015-01-20:
GCode for a STL file printed at 0.3mm creates unwanted Z jump and Y jump.
(My Y plate even went further than print area and belt skipped)
This does not happen at 0.2mm. Nozzle is 0.4mm

See what happens at Layer 17 (Both STL and GCODE)
http://dl.free.fr/kVHGqKvCd

nallath, 2015-01-21:
The image doesn't work.

fcartegnie, 2015-01-21:
That's a file hoster. Not image.

fcartegnie, 2015-01-21:
Only happens on the 1st rendering.

daid, 2015-01-22:
The file hoster doesn't work in our office.

fcartegnie, 2015-01-22:
https://www.dropbox.com/s/z0f4bcgmj545m3l/battery.tar.gz?dl=0

BagelOrb, 2015-04-14:
Your shell thickness should be an integer multiple of your nozzle width!
However, I doubt whether that is the problem at hand.

BagelOrb, 2015-04-14:
Problem is still there on Cura 15.02.1
Can't reproduce the problem in PinkUnicorn.