daid/LegacyCura #1104: Cura 14.12.1 not loading on Windows 8.1

Algomorph, 2015-01-21:
System: Windows 8.1
Cura: 14.12.1 from https://software.ultimaker.com/

Symptoms: trying to start up Cura, it just hangs at the load screen. As soon as I click the load screen, it exits (crashes?).

daid, 2015-01-22:
Do you have proper OpenGL drivers installed?
(As I have multiple people in the office running Cura on Windows 8.1)

Algomorph, 2015-01-22:
What passes as proper? I've got close-to-the latest NVidia driver, I can let you know the version if necessary.

Algomorph, 2015-03-04:
Bump - Cura 15 also doesn't load with the same problem. One thing that's not "typical" about the machine is that the system drive is "S", not "C". Everything else seems typical.

daid, 2015-03-04:
15.02.1?

Algomorph, 2015-03-04:
Correct.