daid/LegacyCura #1106: Bizarre Under Extrusion

InnovationSystemsAU, 2015-01-22:
I'm having some serious issues which have come up on 14.12 with a really bizarre under extrusion. Previously I've had no issues making these parts, and the under extrusion only happens in some circumstances. As you can see, the rest of the part is OK, it's just on this smaller section.

Things I've done to try and resolve it -

1. Played with retraction settings out to both extremes
2. stripped/cleaned the hot end/extruder
3. re-done the e-steps etc, all the standard calibration stuff.

Any ideas? I've really just got it down to potentially a pathfinding issue in the slicer or resume of extrusion bug.

[image]

fluxline, 2015-01-22:
seasonal changes causing the environment your to change, colder?
older filament?

did you update the firmware with the new software?
test with the older software if you get the same problem?
if no, then you will need to find new settings that work with that print,
is my guess

nallath, 2015-01-22:
There are too many variables to say anthing about this. Unless you can prove that it's caused by a certain version (which is entirely possible), there is little we can do.

Try to print the file twice; once sliced by old version of cura, once by the new version.

illuminarti, 2015-01-22:
Try turning off ‘enable combing’ in the retraction settings… or just look at the path in something like Repetier Host.  I suspect that the head may be moving a long way inside the C shaped part to get to the smaller piece, without retracting, and so oozing out a bunch of filament along the way, so that when time comes to print the smaller piece, the head has too low pressure.

InnovationSystemsAU, 2015-01-22:
Thanks Illuminarti, I'll give that a go!

closed 2015-07-06 by BagelOrb