daid/LegacyCura #1108: possibly unwanted behaviour of dividing planes / planner

m-vo, 2015-01-28:
I just noticed a weird behaviour of the head moving lots of times while printing the (solid) bottom layers. The following model was printed (oriented 45°, making the infill pattern perpendicular/parallell to the edges):

[image]
[image]

However the 'infill planes' were not printed at one shot but divided into many parts. Look at this video to see what I mean: http://youtu.be/abjTPHtfon8 - at around 1:00 the size of the parts can be seen.

(I was also wondering why the planner then did not pick the neighbour part as the [nearest] next one?)

daid, 2015-01-29:
Which version of Cura?

m-vo, 2015-01-29:
Its Cura 14.12.1.

(Retraction enabled / Combing enabled / Platform adhesion type: Brim / Fill Density 24% / Plugins: Pause at Height)

daid, 2015-01-29:
Can you verify if the RC for the next release still has this issue?
http://software.ultimaker.com/Cura_closed_beta/

m-vo, 2015-01-29:
From the layer view it looks like it is the same with both versions. But I think I would have to print it to see what is really going on. I'll use the RC for my next prints and let you know if I experience something like that.

daid, 2015-01-29:
If the layer view looks the same, then the same thing is most likely happening.

Any chance you can share the model (you can email it to d.braam@ultimaker.com) then I can take a directly look at it, maybe figuring out why this isn't behaving as it should

JustAnother1, 2015-01-29:
@m-vo What I see in the Video is that he actually takes the closes neighbor part to fill next. The algorithm just does not take into account the avoidance rules(filling the parallel bar is close, but the head moves to the corner and back to not cross the space between) It also does not seem to care in which sequence the part was filled. It therefore picks parts that are closest to where it started to fill this part instead of closest to where it finished filling.

With these limitations in mind. Would that explain what you see?

m-vo, 2015-01-29:
@daid - I just mailed you the model file.
@JustAnother1 - Yes, you could be right and therefore especially the enabled combing is what results in the long movements. But I think the main concern is why those bars get divided in first place?

OT: Btw., I like those updates in cura 15.01 :-), great work!

JustAnother1, 2015-01-29:
The bars are divided in the STL file. It must be like this due to the complex features above the bars. The question is if the slicer can detect in each layer that these segments form a straight bar or not. Due to rounding and tolerances it could be that this is not obvious to the slicer and it therefore does not combine the segments.

m-vo, 2015-01-31:
I see - however I am wondering how this geometry got in there. The part is constructed with some very basic extrudes (in Creo Parametric 3.0). Maybe the bars (which are not regular to the other geometry aren't they?) occur to some odd rounding as well (when exporting to stl). But to be honest, at the moment I don't know the internals of an stl file and how it's constructed.

So if you are sure about the file beeing the problem, should I close the issue? Or should Cura be able to handle these somehow corrupted files as well? :-)

JustAnother1, 2015-01-31:
@m-vo a STL File describes the surface of the model by using triangles. Due to the construction of your model I assume that is uses a lot of triangles along the side of the bars.

The slicer then slices through the triangles and gets a 2d area with lines. The lines are the cuts through the triangles. These small lines then form the bar on that layer.

So I did not assume that there is anything wrong with your model. And yes Cura should be able to deal with this.

Let's see what @daid has to say once he looked into your model.

m-vo, 2015-01-31:
@JustAnother1 Thanks alot for your explanantion. That makes perfect sense.

BagelOrb, 2015-04-14:
Please look at the X-ray view to see whether your model is correct. Furthermore, the model might look correct, but may have double internal faces, meaning it consists of separate parts, which would cause the parts to be printed separately.