daid/LegacyCura #1109: displaced brim line

m-vo, 2015-01-28:
Sometimes I get this effect when printing a brim around an object:

[image]

One line of the brim (usually the first one) is printed a bit further away. This seems not to be reflected in the layer view of Cura (I hope I am right - i cannot reproduce it right now). So this might indicate a firmware problem?

If the object is placed at the very end of the buildplate (like in the image above), the printhead (or probably the black blocks around the axes) seem to crash/touch the endstops. I have not seen that but heard strange noises that (in my estimation) must have occured when the outermost position was reached.

daid, 2015-01-29:
Which type of printer are you using?

m-vo, 2015-01-29:
Sorry, I forgot to mention: Its on an UM² / Cura 14.12.1.

m-vo, 2015-02-05:
Ok that one is probably related:

[image]

* UM² / Cura 15.01 & current firmware installed on printer
* The GCode of the print is available here: http://goo.gl/GkTxLr

What happened: The print was (unfortunately) oriented at the front edge of the buildplate. After some brim lines were printed (very close to the edge) suddenly a displacement happens - this time clearly visible: not only the next brim lines are affected but the whole print. As if the first brim lines and the rest of the object are not aligned. In Cura's path view it looks fine however. So my therory: When printing steps were missed due to the black block reaching the end of the axes (mechanically). So everything gets shifted... Might this be?

(Is the 'security' zone maybe a little bit too short`- or is my printer somehow not calibrated?)

daid, 2015-02-06:
That indeed looks like the head hit a mechanical limit, which could indicate that one of the endstops is not exactly where is should be.

How old is this machine? During the first production run we had some more variation in endstop positions then later machines.

m-vo, 2015-02-06:
Hmm.. The machine is quite new (12/2014). Is there a way I can check the positioning? The endstop switches look ok to me, but perhaps one of them or the board has a problem..? 

That would perhaps explain https://github.com/daid/Cura/issues/1119#issuecomment-72552832 as well.