daid/LegacyCura #1111: Raft, raft, thicker raft!

nothinman, 2015-01-29:
Firstly, let me say something off-topic. Cura is awesome. I've been a huge fan of slic3r, been tuning my printers for years, and just changed the slicer to Cura recently and discovered that my prints are actually perfect and there's nothing left to tune. So thank you for your great work @daid! Well impressed.

And now on topic:
Raft is really good, and works beautifully... for smaller objects.
I tried printing an object that's 310mm x 270mm x 150mm, but unfortunately due to only two layers of raft, and 4 interface layers it started to lift up and warp. 
There is an option to increase the number of interface layers, but not the raft layers.
Why? I would love to print like 20 raft layers, and the rest on top of that.
Is there a way to do this?
and if not:
Can there be a way to do this?

Thanks,
Marcin

quillford, 2015-03-02:
You could add a raft like structure in CAD software like 123D Design. Then you wouldn't need to do it in the slicer.

BagelOrb, 2015-03-03:
20 layers base? That sounds like a lot!

I dont think the warping occurs because of too little layers. I think you need better adhesion to the bed. You could try making the raft base have more lines, less spacing etc. to increase the contact surface with the bed. Also standard methods such as blue tape, glue or alcohol might help in adhesion.

BagelOrb, 2015-04-14:
Why do you want many raft layers?

closed 2015-04-14 by BagelOrb