daid/LegacyCura #1114: Running multiple instances of Cura simultaneously

zohargolan, 2015-02-01:
Hi All,

I am trying to find a way to run multiple instances of Cura simultaneously. lots and lots of times I have both my printers printing in the same time. When I am opening the first instance of Cura, it is reading the configuration file correctly and everything works normally. When I am opening the second instance Cura runs the "first run wizard". Here is what I think happening:

Cura is trying to open the same configuration file for read / write access. Since the file is already open in the previous instance, Cura will not be able to open it, and will run the "first run wizard". In this case I need to configure the printer from scratch. After I am doing that I am printing with both printers simultaneously with no problem. When the prints finish, Cura is trying to write back to the configuration fie, but since its open by the other instance, Cura crashes and locks down.

I tried to install two instances of Cura in different locations, but it didn't help because Cura  uses the User directory which is common for both installations. It would be nice if Cura would use a local configuration directory.

Is there a setting in Cura which will allow using a local configuration file?

Is there any other way to do that?

I was thinking also on virtual machines. Is there a virtual machine that will let me run an application in isolation without installing the whole operating system inside the virtual machine.

Any comment will be appreciated

Zohar

daid, 2015-02-02:
Which OS are you running Cura on?
(You can run different versions side by side without any issues. The same version might be problematic indeed)

zohargolan, 2015-02-02:
I am running on Windows 7, and as you’ve suggested, I am currently running two different versions. Buy I actually need to run 3 different printers simultaneously, which will push me into old versions of Cura.

Best regards,
Zohar Golan
Gol-Tec
zohar@gol-tec.com
www.gol-tec.com
(408)718-2851

From: daid 
Sent: Monday, February 02, 2015 6:35 AM
To: daid/Cura 
Cc: zohar golan 
Subject: Re: [Cura] Running multiple instances of Cura simultaneously (#1114)

Which OS are you running Cura on?
(You can run different versions side by side without any issues. The same version might be problematic indeed)

zohargolan, 2015-02-13:
Hi Daid,
I want to try and work a solution for this feature myself, and pull it if succeed. I've forked the 15.01 tag, but the installed application does not match to the forked code. Is there a different tag I need to fork?
Zohar

quillford, 2015-03-10:
You can use different hosts. Have one printer on Cura, one on Pronterface, and one on OctoPrint for example. If you want to keep Cura available for slicing, just use a third host like Repetier Host.

BagelOrb, 2015-04-14:
When you run Cura from the command line you can run multiple instances of the same version of Cura at the same time.