daid/LegacyCura #1115: Cura not responding after long print

augustjokela, 2015-02-01:
After a print longer the 3 or so hours Cura is unresponsive. I have to do a force quit to shut Cura down. After restarting Cura my machine settings are gone. If I had two machines setup the second one is gone. If I had only one machine setting I have to go through the setup wizard before doing a print. I have experimented with this issue and even deleted my first machine setting. After a force quit, machine 1 is there, machine 2 is not. I recreated machine 2 and delete machine 1. After the next force quit machine 1 is back and machine 2 is gone. This happens even if I restart Cura and my computer multiple times between having to do a force quit. Also the "Welcome to the new version" window appears after a force quit.

I am running Mac OS 10.10.1 thought it happened on the previous version as well. I am running Cura 14.12.1 but this has also happened with 14.09 and 14.07.

InnovationSystemsAU, 2015-02-02:
I can confirm this. Cura is unresponsive after approximately 6 hours of operation. The print continues but it seems that python just hangs. In any case you can no longer control the interface.

matt448, 2015-03-09:
I'm running Cura version 15.02.1 on Mac OS 10.9.5 and I am having this issue as well. Cura becomes unresponsive (spinning beach ball) after a long print and I have force quit the application. When Cura is restarted after a force quit it goes through the first time setup routine and the previous setting have to be recreated.

EdwardHague, 2015-04-13:
I can confirm this - it happens to me after about 3 hours of printing. Not only does Cura use up lots of CPU time, and stop responding to user input, but the ongoing print becomes very slow and intermittent. (prints in ~1 second bursts with pauses in between)

Emanuele-Spatola, 2015-05-19:
I have the same issue on OSX.
I this is a critical issue as the process is using almost 100% CPU for hours, this cause overheating of the CPU and can damage the laptop...

daid, 2015-05-20:
@Emanuele-Spatola if overheating the CPU damages your laptop you should go to Apple, as that should never happen.

The process stuck on OSX is a major issue, but we've done a whole new implementation for this code. And we are close to releasing it.

(PS, would have been nice to report which printers are used, and which type of printing, SD/USB)

Emanuele-Spatola, 2015-05-20:
Thanks for your quick answer @daid.
I'm using a printrbot metal simple Rev F, and I was printing over USB cable.
Let me know if you need further details.
Regarding the overheating, I'm just saying that high temperatures reduce the lifetime of internal components, that's all.

Jnesselr, 2015-06-22:
Any update on that newer code? I was curious as to why the serial communication was split out into another process. I believe the issue alephobjecta/Cura#84 is related to this. I can build on Mac OS X so I can tested changes when needed.

nallath, 2015-06-23:
This code is going to be depricated. The new Cura can be found at https://github.com/Ultimaker/cura. I've updated the USB printing code for that version, so it should be much easier to work with it.