daid/LegacyCura #1119: printer unresponsive after firmware update

m-vo, 2015-02-02:
What I did:

* installed the default firmware that comes with Cura 15.01-RC11
* selected MATERIAL / CHANGE on the printer, then ABORT (no new material)
* removed the USB cable
* selected MAINTENANCE / MOVE MATERIAL and turned the knob a bit

What then happened:

* feeder motor started and kept continuously spinning (even after exiting from the menu option)
* printer not respondig to commands from the menu (like home head)

After power cylcing everything works fine again, so that might correspond to the firmware update? Not a big thing, but perhaps it helps.

m-vo, 2015-02-02:
...followed by that one short after (when inserting new material). I don't know if there is a link between these two issues.

[image]

(That one was gone after power cycling as well. Tried a print after it which suceeded.)

daid, 2015-02-03:
The RC11 still had some minor issues, you should upgrade to the 15.01 final
http://software.ultimaker.com/

I'll have a go in seeing if I can reproduce your first issue with the motor keeping running.

The X/Y switch stuck error happens during homing, if the switches are still pressed after backing up 7mm after hitting the switch.