daid/LegacyCura #1121: Nozzle not heating on start - Cura 14.12.1

jdmech, 2015-02-04:
Hi folks - 
For an unknown reason - the latest version 14.12.1 of Cura is
having some sort of gcode writing glitch when saving to a sd card.
When the print is started, the nozzle temp reads 0.
I have uninstalled and reinstalled the previous version - 14.09. 
Earlier version seems to run normal. Hope this helps - 

OS: Windows 8 - 

cheers - jdmech

daid, 2015-02-04:
Which printer? Happens every time? Any plugins enabled?

jdmech, 2015-02-04:
Frequency: Once the problem starts, it repeats every time.
Not sure what causes it.

My printer and setup:
Prusa I3 - Rework (home built)
w/ heated print bed. Borosilicate glass plate.
Ramps 1.4 board, Arduino 2650
Thermaltake 750watt power supply
Ulti-panel lcd controller
Marlin firmware
Cura - slicer/print software.

Plugin enabled: TweakAtZ.

daid, 2015-02-04:
Ok, first, upgrade to 15.01, as this fixes some problems with plugins.

Can you check if you save the GCode, if there is a M109 command in there?

jdmech, 2015-02-04:
Great! -  will try the upgrade tonight.
I'll also take a look at the gcode file and give you feedback.

Thanks for the quick response Daid - cheers - jdmech

closed 2015-02-17 by jdmech