daid/LegacyCura #1127: Not able to assign parts to intended extruder

camchaney, 2015-02-05:
I have just started to try out my dual extruder on my Lulzbot Taz. I uploaded the basic configuration file (http://devel.lulzbot.com/TAZ/accessories/dual_extruder/dual_extruder-1.0/Calibration/calibration_squares/Calibration_Squares.amf.xml) into Cura with no problems. The red and yellow colors (representing which extruder to use) of the model were in the right place and everything looked fine.

I started to think, though, how I would be able to re-assign the colors if I wanted to change the extruder that printed each part. I looked all around in Cura, but there was no way of assigning an extruder to a specific part of my model. Once you upload an amf file, or combine two parts with dual extrusion merge, there is no way of changing the extruder that they are assigned to.

This problem is really something that they should take care of in the next update. There should be some sort of option, when you click a part, to be able to change the extruder in which it is assigned.

daid, 2015-02-05:
This will come in the GUI overhaul that we're working on.

closed 2015-02-05 by daid

daid, 2015-05-03:
Counterbump!

XiaoShiZi, 2015-05-03:
Playing pinball?

BagelOrb, 2015-05-03:
Pinball might be a feature of the new GUI as well, who knows?

francoisba, 2015-05-18:
Hello, similar feature request here (https://github.com/daid/Cura/issues/675)
I add my vote ! ;)

Many thank to the developers

emosuka, 2016-02-03:
So is this going to happen?

nallath, 2016-02-04:
We don't have it yet, it is planned for a feature release.

BagelOrb, 2016-02-04:
You can already assign an extruder to each object separately, but dual extrusion is not fully supported yet by the new Cura.

There are a lot of settings which you can change per object now! :D

francoisba, 2016-02-13:
> You can already assign an extruder to each object separately

Hello everyone, thank you for your answers.
@BagelOrb, could you please tell us a bit more about that, please ?
Because I didn't find out how to do it. On which version of Cura is it possible ?

Thank you for your help

BagelOrb, 2016-02-14:
Thats actually on cura 2.0, which hasn't fully been released yet. You can
find a beta online which I think is called 15.10, but it is from a while
back and has quite some bugs in it.

Me saying that that was possible on an issue in this repository was kind of
misleading, since the repo for that cura is at Ultimaker.github.com/cura