daid/LegacyCura #1133: Toolpath to SD is not functional anymore in Cura 15

tom111, 2015-02-10:
I just upgraded to the new version of Cura and the "Toolpath to SD" is now disfunctional. It correctly switches between Save and SD when the card is inserted/ejected, but it does not do anything when trying to save to SD. Saving to disk works fine and it worked fine with any version of Cura 14.
This is on Mac OS X 10.10.2.

ticino80, 2015-02-11:
Hi tom,
I have the same problem
My system: Mac OS X 10.10.2
MacBook Pro 15" early 2011

pmsimard, 2015-02-11:
I can't reproduce this issue in OSX 10.9.3.
http://umforum.ultimaker.com/index.php?/topic/9595-toolpath-to-sd-cura-1501/

daid, 2015-02-12:
Got quite a few reports of this happening in MacOS right now. The workaround seems to be to empty the new "replicate folder structure to SD" setting in the preferences.

wyojustin, 2015-02-12:
I had the same problem on Ubuntu.  Work-around was to right-click save-gcode as.

dwardio, 2015-02-12:
Same thing occurs in OS X 10.9.5.  Can use the save as workaround, but the button doesn't do anything.

pmsimard, 2015-02-12:
Daid,

I updated my OSX and it turns out its a permission error when creating the directory on the sd card.

Creating replication directory: /CuraExamples
Traceback (most recent call last):
  File "Cura/gui/util/openglGui.pyo", line 184, in _OnGuiMouseDown
  File "Cura/gui/util/openglGui.pyo", line 99, in OnMouseDown
  File "Cura/gui/util/openglGui.pyo", line 548, in OnMouseDown
  File "Cura/gui/sceneView.pyo", line 268, in OnPrintButton
  File "os.pyo", line 157, in makedirs
OSError: [Errno 13] Permission denied: '/CuraExamples'

Muriukidavid, 2015-02-13:
Same problem in Ubuntu 14.04 , I use File->Save Gcode menu

pmsimard, 2015-02-13:
Plenty of work arounds already in the comments and link above. No need to add more reports.

daid, 2015-02-13:
Build a 14.02-RC1 which I hope fixes this issue. Can someone test to see if that works?
http://software.ultimaker.com/Cura_closed_beta/

MiguelAngelLV, 2015-02-16:
Same problem with Debian Jessie 64 bits.

WZ9V, 2016-05-22:
This is working for me in Linux Mint 17.3 64-bit using Cura 15.04.5