daid/LegacyCura #1135: Layer view displays very wide lines

MechanizedMedic, 2015-02-11:
Same issue as reported in the forums here: http://umforum.ultimaker.com/index.php?/topic/9537-cura-tool-path-display-issue/

When viewing any model or G-code file the extrusion paths displayed are much wider than they should be. A few layers are still displayed correctly, but I could not find anything that would cause the layers to be differentiated like this.

[image]

daid, 2015-02-12:
No clue. Which printer? Which configuration?
(The material flow setting causes this for example)

MechanizedMedic, 2015-02-12:
It's for a PrintrBot Simple Makers Kit. The configuration is very standard fare, the only change is print dimensions. The first layer extrusion setting is set to 150% but the nozzle is .4 and flow is 100%.

It there anything that I can do to help debug this?

daid, 2015-02-13:
Still not seeing the same result...

Can you send me a copy of the setting used, as well as your machine settings?
On windows you can find these as ini files in:
C:\Users\[user]\.cura\15.01\

MechanizedMedic, 2015-02-20:
Well, I just upgraded to 15.02.1 and the problem is now gone.

closed 2015-02-20 by MechanizedMedic