daid/LegacyCura #1136: Cura 15.01 not loading on Windows 8.1 or higher

jason13168, 2015-02-12:
System: Windows 8.1
Cura: 15.01 from https://software.ultimaker.com/
Symptoms: trying to start up Cura, it just hangs at the load screen. As soon as I click the load screen, it exits

daid, 2015-02-12:
Can you post the contents of the log file. At:
C:\Users\[user]\.cura\15.01\output_log.txt

jason13168, 2015-02-12:
C:\Program Files (x86)\Cura_15.01>cura.bat
Traceback (most recent call last):
  File "C:\Program Files (x86)\Cura_15.01\python\lib\runpy.py", line 162,
in _run_module_as_main "__main__", fname, loader, pkg_name)
  File "C:\Program Files (x86)\Cura_15.01\python\lib\runpy.py", line 72, in
_run_code exec code in run_globals
  File "C:\Program Files (x86)\Cura_15.01\Cura\cura.py", line 13, in
<module> from Cura.util import profile
  File "Cura\util\profile.py", line 501, in <module>
setting('sdcard_rootfolder', _getMyDocumentsFolder(), str, 'preference',
'hidden').setLabel(_("Base folder to replicate on SD card"), _("The
specified folder will be used as a base path. Any gcode generated from
object coming from within that folder will be automatically saved on the SD
card at the same sub-folder. Any object coming from outside of this path
will save the gcode on the root folder of the card."))
  File "Cura\util\profile.py", line 57, in __init__ self._default =
unicode(default)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xb3 in position 9:
ordinalnot in range(128)

qoob, 2015-02-12:
Can confirm this, it happens when there are non-ascii characters in MyDocuments path(most probably when a username contains them, or a locale is not US-en). I just changed path in util/profile.py, line 493 to ascii-only string as a workaround.

jason13168, 2015-02-13:
changed path in util/profile.py, line 493 to ascii-only
is success
thanks

daid, 2015-02-13:
Applied a workaround, as this cannot be a 100% fixed. But it will no longer crash on startup (will roll out update soon, also due to the SD MacOS issue)

closed 2015-02-13 by daid