daid/LegacyCura #1137: Cura 15.01: hotend and heatbed don't heat simultanously anymore?

freeewilly, 2015-02-12:
Before (14.9) Cura heated hotend and heat bed at same time, now it heats the bed first and then the hotend, making it much slower until the printer is ready to print...

daid, 2015-02-12:
This is intentional, as most printers have a bed heating time of 5 minutes, while the hotend heats in 1 minute. This mend the hotend was sitting hot for a long time, and dripping empty.

You can change this behavior if you make custom start code. Just read the comments in the start code.

closed 2015-02-12 by daid

freeewilly, 2015-02-12:
Got it! Thank you and sorry for my ignorance.

fhuberts, 2015-02-12:
daid, you could also keep track of those times: the first time you heat serially and remember the times. the next time(s) you heat in parallel, using the previous times and also remember the times

daid, 2015-02-12:
No problem. There are loads of different configurations, the defaults are tailored towards the most common ones. But there are lots of possibilities, sometimes people just need to be pointed to the right place to know what to adjust.

daid, 2015-02-12:
@fhuberts I could, but I won't.