daid/LegacyCura #1138: Exception while printing

kennedyshead, 2015-02-12:
Unexpected error while reading serial port: SerialException: 'device reports readiness to read but returned no data (device disconnected?)' @ machineCom.py:_readline:516
Changing monitoring state from 'Printing' to 'Error: SerialException: 'device reports re...'
Connection closed, closing down monitor

Gcode works fine in other software

daid, 2015-02-13:
Sadly, this exception comes from the driver and there is nothing I can do about it.

closed 2015-02-13 by daid

kennedyshead, 2015-02-14:
Oh I was not aware, well... I wonder what causing it :/ Will dig in FW

daid, 2015-02-14:
Firmware won't help you here, it's the windows drivers that are messing things up.

kennedyshead, 2015-02-14:
Aha Im on debian ;) ill try a later kernel