daid/LegacyCura #1140: Can't see dimension and time of print

Simmonsstummer, 2015-02-12:
Hi, in the version 15.01 i can't see dimension in the resize button and the time of print.

restinpieces, 2015-02-12:
Same for me. When I load something the text and the bar flickers between white and black. and after it finished calculating its gone. Also the text one the saved file name notification is invisible.

daid, 2015-02-13:
No idea what's going on here. The rendering code hasn't been really touched at all. But more people are reporting it in 15.01 on the UM forums..

stefilla, 2015-02-19:
I have installed the version 15.01 form mac and i can't see dimension in the resize button and the time of print. I  cant change the support type and platform. I try reinstall the old cura but there is the same problem.

daid, 2015-02-19:
There is a 15.02.1 which fixes this issue now.

closed 2015-02-19 by daid

stefilla, 2015-02-19:
Thank you, but I tried to install the upgrade version but I have the same problem on Mac OS system but on windows run well.

Stefania - 3D StudioLab

vishnubob, 2015-04-15:
I still have this problem on OSX, running Cura 15.02.1.