daid/LegacyCura #1142: 15.01 - Some parts of model skipped

quillford, 2015-02-15:
I'm slicing this but not all of the model comes out. Some of it is skipped. [Model](http://www.thingiverse.com/thing:215010/#files) in question
OS X: 10.10.3
[image]
[image]

daid, 2015-02-15:
Does the model show any red in xray view?

quillford, 2015-02-15:
No. 
[image]

BagelOrb, 2015-04-14:
What's skipped?
Are you talking about the thin walls?
Perhaps they are thinner than twice the nozzle size. In that case they cannot be printed because they are too thin.

closed 2019-09-17 by quillford