daid/LegacyCura #1143: Stuck process grabbing 99.8% CPU on OSX

NetForces, 2015-02-15:
Cura 15.01 on OSX Yosemite 10.10.2

I think it's when you stop/abort a print that the process get stuck.

[image]

lerouxb, 2015-02-19:
I had the previous version of Cura on OSX crash after long prints, then I had to "force quit" it, then it left a process taking 99%+ CPU, even though the OSX gui thought the app was closed. Only noticed after the third one. Only way to kill it was from activity monitor.

Then when I restarted it after that happening it ran through that "choose your printer" wizard again every time.

I haven't seen it happen with version 15 yet.

mikehedman, 2015-05-11:
I believe that this issue and #1115 are probably the same issue.  (and one that I am also experiencing with 15.02.1)