daid/LegacyCura #1144: Cura does not stop extruder heat when aborting/stoping a print

NetForces, 2015-02-15:
If you abort/stop a print the extruder temperature stays at the print temperature.

quillford, 2015-03-02:
Are you printing with Cura? If you are just slicing, then that is an issue with your host. The end gcode isn't run because you aborted the print before it could get to it.

NetForces, 2015-03-02:
Slicing and printing to a Printrbot Simple.

Very simple to reproduce.

1. Start a print
2. Wait for the print to actually start, maybe wait for 1 layer or 2.
3. Press the Cancel button in Cure. After a few seconds the print will stop and the head will not be moved and extruder temperature is not stopped (stays at whatever you has for the extrusion temperature).

quillford, 2015-03-02:
Ok. So you are using Cura as a host. When you abort a print, the end gcode doesn't run therefor the command for turning all heaters off does not run.

NetForces, 2015-03-02:
I think it would be important to allow a configuration to run the end gcode on a abort. Or else someone may leave his printer with the heaters on without knowing it.

For exemple Repetier on an abort does not run the end gcode, but does turn off heaters.

tebruno99, 2015-03-16:
I agree completely, this has actually happened to me the first time I used Cura 2 days ago.  Not only that but then I couldn't figure out how to get Cura to move the head so I could reset everything.