daid/LegacyCura #1148: Nozzle not heating on start - Cura 15.01 for Win 7 (32bit)

jdmech, 2015-02-17:
Ok maybe this is user error (and please point me to where I have overlooked something) but the bed appears to start heating while the extruder nozzle stays at level 0 deg/C. After the bed reaches it's target temp, THEN the nozzle starts heating up to the programmed target temp (200 deg/C) Is this a new feature? It did the same on my Windows 8.1 (64bit) machine with the previous version of Cura - 14.9. I saved the cgode if you would like to see it.

Thanks! - jdmech

[image]

daid, 2015-02-18:
It's a new, and intended, feature. If you want to do things differently, look at the start code, you can uncomment 2 lines there, if you add those, you can do everything in the order you want, and Cura won't add it's own temperature lines.

closed 2015-02-18 by daid

jdmech, 2015-02-18:
Ah - Ok that's great - I'll stick with the intended feature. You guys are
doing a fantastic
job with developing and updating Cura.

Thanks for your help! - jdmech

Regards,

Jon Dawe (JD)
jdmech@gmail.com