daid/LegacyCura #1152: Scaling should have at least 3 to 4 places after decimal point

reputech, 2015-02-19:
When printig with PLA, shrinking due to temperature is usually about 0,6 to 0,7 %
So I upscale my parts with a factor of about 1.0065
this is shown rounded to 1.01 so you can't see anymore the real scaling factor.
The issue is only the display of the scaling value. The measures are correct calculatet in all directions.

quillford, 2015-03-02:
For the time being, you could use something like NetFabb for scaling.

reputech, 2015-03-02:
Scaling works perfect in Cura! Even if I put 1.0065 as scale.
The problems is only that i cannot see the scale value, because it's being rounded do 1.01.

daid, 2015-03-03:
I've updated the scaling to 3 decimals for the next version.

closed 2015-03-03 by daid