daid/LegacyCura #1155: Profiles erased upon updating

quillford, 2015-02-20:
I have noticed this on every Cura version I have used (since 14.07). My profiles themselves won't be erased but the contents will be restored to the generic default. Only the name of the profile is retained. I have noticed this on OS X (10.9 & 10.10) as well as Windows 7.

daid, 2015-02-20:
No idea why that's happening, the profiles are copied over for each new version and simply loaded...

mildmojo, 2015-02-20:
This just happened to me when I upgraded from 15.01 to 15.02.1 on Linux. I didn't realize at first, and the default Cura `start.gcode` crashed the hot end on my delta because it was missing my required bed leveling gcode. I caught it before it could shatter the acrylic arms. :relieved:

Nutz95, 2015-02-25:
It's systematic on my side for every updates.
The general machine settings stay the same but the default settings are applied on all Basic and Advanced  panel.
Also my start and End-G-Code changes are restored to default.

I reproduce it on both Linux 64bits and windows 32 bit
Any idea?

TTN-, 2015-02-28:
+1 same here. Very frustrating. Upgraded using the .deb package on ubuntu 14.04

mdmbc, 2015-03-07:
I've seen the same thing on linux every time I update, but they're not actually erased. They're just not copied over to the current version's config folder. You can copy them from the old ~/.cura/[version] folder to the new one, then everything works as expected.

TTN-, 2015-03-08:
Well the profiles certainly aren't copying when updating for me. So this is definitely a bug. I'm running on linux btw.

the-edge01, 2015-03-09:
I have the same problem.  The machine settings copy over, but the individual print defaults do not copy.  Using windows 7 x64.

Nutz95, 2015-04-11:
I have more details about this issue:
The problem seems to manifest when you have "custom" profile.
On update, the custom profiles are overwritten with the default "custom" configuration.
Does this help?

SparkyCola, 2016-05-28:
I had the same issue on Ubuntu 16.10 when updating Cura from 15.04.4 to 15.04.6. It reset all settings to default with updating and even caused a failed print because the diameter was reset to 2.85 (I didn't realize this error). Workaround would be a warning/message after updating asking you to restore your settings (e.g. from G-CODE).

constructive-realities, 2017-04-23:
Super lame the CURA can't persist a custom printer setting between updates. maybe someday silly will get fixed.  

If you don't want to fix it, how about a notification step and some instuctions on how to restore them.  lost a really good setup today.

TTN-, 2017-04-23:
@constructive-realities do you have a gcode file generated by your good setup? Load the profile into cura from it. All profiles are saved in gcodes generated by it. It is how I restored mine.

closed 2019-09-17 by quillford