daid/LegacyCura #1156: Unnecessary retracts

igi79, 2015-02-20:
There is something wrong with new slicing engine in version 15. Same model with same setting is estimated for 1h41m in version 14.09 but in 15.02.01 it is 1h46m. There are nearly 3.5x more retracts - 175 in version 14 vs. 602 retracts in version 15. With same settings!

daid, 2015-02-20:
Need like, a ton of more information then this. (But possibly it's the same problem as we're working on on the ultimaker forums right now)

jaeg, 2015-02-21:
Here's a picture of what one of my layers look like between 15.02.01 and 14.09 if this helps:
http://imgur.com/8aJJgDx

orangeudav, 2015-03-23:
Maybe same problem, i see retracting for every end of infill lines, between infill lines

BagelOrb, 2015-03-25:
Do you guys have the same problem in the newest version of Cura?
Please download, install and verify.

jaeg, 2015-03-25:
Is 15.02.01 still the newest?  That's when I first started having the issue.  That's also the version I used to make my image above.

tecan, 2015-03-31:
when the disable retractions check box is off it still is enabled.