daid/LegacyCura #1158: weird behaviour after mac resume

rooiejoris, 2015-02-22:
Hi,

There is a weird behavior after resuming my mac, but very very consistent.
After resuming it looks like the cura thinks the shift key is pressed. I noticed first because rotating the view is not possible anymore then, it is just panning [very annoying!]. And when you want to rotate an object it doesn't snap every 15 degrees.

now i always have to restart cura after opening my mac screen

[not high priority...]

cheers \ joris

PyroNyzen, 2015-06-26:
I get this issue too. I think it's something to do with putting the mac to sleep and waking it.

daid, 2015-06-26:
It's a problem with wxPython. Atleast now we know what causes it (I do knew the shift key could get on "always down" on mac)

It is however, a "won't fix". For 2 reasons, we're working on a new version based on pyQt instead of wxWidgets, and it's a bug in wx which I'm not going to dive into for this bug.

PyroNyzen, 2015-06-26:
Have a look at this code example, this should fix the issue if it can be
implemented into the main program. http://pastebin.com/vni44Gs6
Basically you need to bind to a couple of other events, specifically
wx.EVT_KEY_DOWN and wx.EVT_KEY_UP so the program knows when it is up or
down.
Or call that check to be run when switching in between view modes.

daid, 2015-06-26:
But I'm not looking at the events (as you will only get events during focus, which will cause different bugs)
https://github.com/daid/Cura/blob/SteamEngine/Cura/gui/sceneView.py#L757

PyroNyzen, 2015-06-26:
The issue with the shift key only affects mouse interaction, Panning.
Isn't wx.GetKeyState() asynchronous to the mouse event?