daid/LegacyCura #1159: Feature request: Option to change print ordering of infill vs. perimeter.

mosh1, 2015-02-23:
v15 changed the ordering from printing perimeters first to printing infill first. This can cause bad surface artifacts as the perimeter shrinks around the printed infill. An option to choose this order would resolve the issue.

More detailed info of surface quality issues: http://umforum.ultimaker.com/index.php?/topic/9771-1501-major-surface-quality-issues/

mosh1, 2015-02-23:
I'd be happy to implement this change if someone can point me where to look in the slicer engine where this is implemented.

daid, 2015-02-24:
https://github.com/Ultimaker/CuraEngine/blob/master/src/fffProcessor.h#L647
https://github.com/Ultimaker/CuraEngine/blob/master/src/fffProcessor.h#L681

Should help

jmeosbn, 2015-03-03:
Hi, was this an intentional change? I was about to report this as a bug.

Additionally, I get traditional ordering (perimeter first) on the top and bottom shells, and the reverse for the infill layers between.

mosh1, 2015-03-03:
According to the thread mentioned at top this was an intentional change.

closed 2015-03-03 by mosh1