daid/LegacyCura #1166: 15.02.01 - Export is missing layers

quillford, 2015-02-28:
When I export the gcode for my model, it will come out missing every layer or so. Sometimes it comes out with only the first few layers exported then the rest blank and sometimes everything is fine. This has happened on multiple models. I fixed the models with netFabb to make sure it wasn't the models. 

Version: 15.02.01
OS: OS X 10.10.3

Model pictured: https://github.com/josefprusa/Prusa3-vanilla/blob/master/distribution/x-carriage.stl

[image]
[image]
[image]

daid, 2015-03-02:
Do you have the layer view open during gcode generation? Cause this messes with the layer view for some users (not sure why)

quillford, 2015-03-02:
No. It wasn't open.

BagelOrb, 2015-03-03:
Is it a problem with the LayerView, or does the exported file actually miss the layers?

quillford, 2015-03-03:
It actually misses layers in the gcode.

N720, 2016-04-19:
I am having a similar problem, and I believe I have traced it to a bug when switching between printers.  In my case it gives mesh-like prints when I switch from one printer to another (Printrbot Simple Metal, Printrbot Metal Plus, and Witbox 2).  My temporary work-around was to uninstall and re-install Cura.

BagelOrb, 2016-04-20:
Another workaround would be to uninstall and install the Cura 2.1 beta ;)

closed 2019-09-17 by quillford