daid/LegacyCura #1168: Paused and the head ran into my part when homing

jsquaredz, 2015-03-02:
I had to 50 mm tall parts on being printed.  First the one on the left was printed, then the one on the right started.  I went in and paused the print, and the head proceeded to go into a material loading position and smash through my first part.  It essentially went up just above the part it was printing rather than going up above the highest part before heading to a resting place.

daid, 2015-03-02:
I assume this is on an UM2?

BagelOrb, 2015-03-03:
That doesn't really matter. It seems to me that for any printer it would be safer to first move up and then to a resting place.

BagelOrb, 2015-04-14:
This is a firmware issue. I assume you paused the print via the printer itself, not via Cura.

closed 2020-05-01 by jsquaredz