daid/LegacyCura #1170: Version 6.3.1. Model viewing issue

Waynicus, 2015-03-03:
i've just downloaded and installed 6.3.1. and when you load a model the only view you can select to see what's on the hotbed is the layer view. All other views don't show anything?

Waynicus, 2015-03-03:
My fault, I stupidly opened the gcode file and not the stl. file

closed 2015-03-03 by Waynicus

daid, 2015-03-03:
There is no 6.3.1 version.

Waynicus, 2015-03-03:
There is for Creatbot. It also has a direct link to the forums.

daid, 2015-03-03:
Well, report the problem to them then. I can only give support for the versions that I release at software.ultimaker.com

Waynicus, 2015-03-03:
OK. First thing, I used their software which links to this forum! Not my
issue! I suggest you contact them if you don't want Creatbot users (which
uses Cura software) to put up any issues.
Second thing, I closed the issue as I know what I did wrong! So there is
now no problem.