daid/LegacyCura #1178: Check for Updates does not work

quillford, 2015-03-09:
I have tried using the Check for Updates button several times when I know there is a new release, but it simply returns that I am on the latest version. 

OS: OS X 10.10.3, Windows 7

daid, 2015-03-09:
The update check looks for this file:
http://software.ultimaker.com/latest.xml

But if it cannot reach it (for example, a firewall blocking it) it will show that there is no newer version. It's working for for me here on Windows 7 (got every old version installed)

closed 2015-03-09 by daid

foreachthing, 2016-04-07:
http://software.ultimaker.com/latest.xml contains wrong information:
`<cura><release os="Windows" major="15" minor="2" revision="1"><filename>Cura_15.04.3.exe</filename></release><release os="Darwin" major="15" minor="2" revision="1"><filename>Cura-15.04.3-MacOS.dmg</filename></release><release os="Linux" major="15" minor="2" revision="1"><filename>cura_15.04.3-debian_amd64.deb</filename></release></cura>`
New release is 15.04.5 and my 15.04.4 thinks it's the latest version.