daid/LegacyCura #1180: Feature Request: Warning for non manifold models

quillford, 2015-03-10:
It would be nice to have a notification if a model was not manifold.

Some ideas:

Warning text - could also be in the form of a pop when the model is initially loaded
[image]

A different designated color for non manifold models
[image]

BagelOrb, 2015-03-10:
I think a warning message is better than another color. In some cases we already use a different color for different models.

However, checking whether a model is manifold might be not as easy as it seems. The good news is that we have most of the machinery in place; we could quite simply make a plugin to do just that.

quillford, 2015-03-10:
What type of plugin would be used? As I understand it, there are only window and post processing.

quillford, 2015-03-10:
Another idea: Include an icon with a warning symbol next to the three buttons that show up when the model is clicked on.

daid, 2015-03-11:
Problem here is that a lot of models are not strictly manifold, but slice just fine. So it would trick people in thinking in something is wrong, while actually nothing is really wrong. At which point people will just ignore the warning all the time, and not trigger on it when it's really needed.

closed 2019-09-17 by quillford