daid/LegacyCura #1185: Cura starts partially off screen on Mac OSX

chuckr, 2015-03-11:
I'm on a MacBookPro with OSX Yosemite 10.10.2 with two heads (Retina and Dell 3014) 

If I start Cura and resize the app to be larger, then quit the app, the next time I start Cura, it starts with the top part of the app off the display area.  In OSX there is no way to grab the app window and reposition it back onto the screen.  The only workaround to get Cura to get back properly on the display is to muck around with display resolutions which will eventually trick it back on the screen.  Screen shot attached.

[image]

matteliot, 2016-12-07:
Same issue.  MacOS Sierra.

Uninstalling then reinstalling didn't work - it kept the positions.

BagelOrb, 2016-12-08:
Try Cura 2.3; it's better in almost every way!!

matteliot, 2016-12-08:
Will do, but I was able to resolve this by starting it then making the window as small as possible by dragging the bottom right corner, then quitting and coming back in.

When it was partially offscreen, it wasn't usable since the buttons on top and the LHS controls were off screen.