daid/LegacyCura #1186: Platform adhesion doesn't work

vguillard, 2015-03-11:
Hello,
I've used Cura for 3months and I love this tool.
However, since yesterday, I can't print with raft and today, I've just discovered I can't print with raft or brim or skirt.
I saved the different gcode : 
  . there's no difference between "brim version" and "skirt version"
  . in "raft version", it's written :
 ;Layer count: 137
;LAYER:-2
;RAFT
;LAYER:-1
;RAFT
;LAYER:-1
;RAFT
;LAYER:-1
;RAFT
;LAYER:0

I don't understand why there's no gcode after the comments

Thank you

daid, 2015-03-12:
Does layer 0 have gcode?

vguillard, 2015-03-12:
Yes,
;LAYER:-2
;RAFT
;LAYER:-1
;RAFT
;LAYER:-1
;RAFT
;LAYER:-1
;RAFT
;LAYER:0
M107
;LAYER:1
M106 S127
G1 F600 E-4.00000
G0 F9000 X78.162 Y78.745 Z1.410
;TYPE:WALL-INNER
G1 F600 E0.00000

It's really strange, I created a windows VM, installed cura and it worked ... until i load my piece and after that, it doesn't work anymore.
After cura fresh-installed, I read the gcode generated for the bot ultimaker and I saw the gocde after the comments raft.
I loaded my piece : it worked.
I removed the bot : it didn't work anymore.

I thought the issues was coming with my piece but on another PC, I did never load this piece and the raft doesn't appear in the gcode.

I'm using Cura 15.02.1 on LinuxMint + Windows 7

daid, 2015-03-12:
Actually, no. Layer 0 is the first layer. And it's empty, which explains the empty raft.

vguillard, 2015-03-12:
Here is a snippet where there is a draft and the layer 0 is empty:

;LAYER:-1
;RAFT
G0 F9000 X118.999 Y97.527 Z1.110
;TYPE:SUPPORT
G1 F1200 X118.999 Y83.407 E88.12370
... <some gcode> ...
;LAYER:0
M107
;LAYER:1
M106 S127
G1 F600 E122.76981
G0 F9000 X95.863 Y94.254 Z1.410
;TYPE:WALL-INNER

daid, 2015-03-12:
But there is also support, and support is also added to the raft. So that's generating the raft then.

You're free to think whatever you want, but I made the code. So I'm quite sure on how it works.

vguillard, 2015-03-12:
No problemo, I may misunderstood how raft works. :-)
However, I don't explain why there is no skirt, no brim and no raft.

I've juste tried with this simple cube: http://www.thingiverse.com/download:17274
Here the links to the gcodes : 
  . with skirt : http://pastebin.fr/38943
  . with brim : http://pastebin.fr/38944
  . with raft  : http://pastebin.fr/38945

The "skirt gcode" and "brim gcode" are similar ... and I can't explain why.

I'm going to be crazy because I really don't know what I did/misconfigured... maybe, I focus on bad things :'(

vguillard, 2015-03-12:
I've just increased the width of the 1rst layer and ... it works now.
So, it was not a software bug, it was a user bug. :-s
I'm sorry.

closed 2015-03-12 by vguillard