daid/LegacyCura #1189: error slicer

lamerhouse, 2015-03-13:
I have a piece that is at the bottom than at the top should be open .. but when it generates the g-code them always closes version 15.2.1 attach files
[image]
[image]

[image]   rename PNG in ZIP contains file stl

BagelOrb, 2015-03-16:
try expert settings > fix horrible > uncheck combine-everything type-A

closed 2015-03-16 by BagelOrb

lamerhouse, 2015-03-16:
perfect tx

daid, 2015-03-18:
Note, that means normals in your 3D model are actually flipped.