daid/LegacyCura #1191: Print missing Z axis hole.

mortnr, 2015-03-14:
I am fairly new to 3D printing, so this may be something obvious.  I drew a part on sketchup, checked it on netfab and imported it to cura.  In each case the model looked fine.  However when I printed it, the print out was missing a major element of the model - namely a hole that ran through the Z axis.  Any ideas where to look for the problem?
Thanks.
Nathan
[image]

Dim3nsioneer, 2015-03-14:
I recommend to check the layer view of Cura before printing (icon top right and then the last option).
Check the expert settings (menu expert); if any of the 'fix horrible' option is enabled, disable it. This usually solved that kind of problems.

btw: Shell thickness should be a multiple of 0.4mm for a standard UM2 hotend (1mm makes sense if you have a .5mm or 1mm nozzle).

mortnr, 2015-03-14:
Thank you. Uncle king 'Fix horrible' solved the problem.
Best regards, Nathan

closed 2015-03-16 by BagelOrb