daid/LegacyCura #1192: Unable to save gcode

cichym, 2015-03-14:
Unable to to save g code, 
Cure doesn't seems to finish calculation, and it isn't possible to save the g code file.

Dim3nsioneer, 2015-03-15:
Any plugin activated? It might be a plugin which is stuck in an endless loop.

guidoklauw, 2015-03-17:
Did you fix it? I'm having the same problem with Cura and it's not caused by any plugin...

cichym, 2015-03-17:
Unfortunately,I didn't fix it. I am using older version...

daid, 2015-03-18:
What does the log file say? (it's in your user directory)

cichym, 2015-03-18:
That was a lesson for me. Problem was in filament diameter, I should use dot "2.96" , not comma "2,96".

Log file bellow:

Listening for engine communications on 49674
Exception in thread Thread-7:
Traceback (most recent call last):
  File "C:\Program Files (x86)\Cura_15.02.1\python\lib\threading.py", line 552, in __bootstrap_inner
    self.run()
  File "C:\Program Files (x86)\Cura_15.02.1\python\lib\threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "Cura\util\sliceEngine.py", line 418, in _runEngine
    self._result.addReplaceTag('#F_WGHT#', math.floor(self._result.getFilamentWeight(0) * 1000.0))
  File "Cura\util\sliceEngine.py", line 74, in getFilamentWeight
    radius = float(profile.getProfileSetting('filament_diameter')) / 2
ValueError: invalid literal for float(): 2,96

Closing down

closed 2015-03-19 by nallath

francoisba, 2015-05-20:
Hello,
I know it's closed, but I have the same problem.
As I'm already using dots, there might be another explanation.

Maybe for you who know this software, another clue : my model reminds grey on the platform, and the layers view displays only one layer (in this mode, my model disappears)

Thank you for your help,

Best regards