daid/LegacyCura #1194: XY Compensation

MiguelAngelLV, 2015-03-16:
My printer make the holes too small (I don't know why) but with Slic3r XY Compensation I solved it.

Would be posible add the feature to Cura?

BagelOrb, 2015-03-25:
That would be fairly simple; just add a distance to all insets.
@Diad would we want an extra setting for that?

BagelOrb, 2015-03-27:
Option has been implemented in CuraEngine. Not (yet?) present in front-end. Perhaps this is not an option which should be exposed to standard users.

MiguelAngelLV, 2015-04-05:
How configure this feature? Could I set it from .ini file?

BagelOrb, 2015-05-12:
You can set it in fdmPrinter.json or via the new Cura interface.

BagelOrb, 2015-05-12:
Or wait for the new Cura to be released ^^

closed 2015-05-12 by BagelOrb

nallath, 2015-05-12:
Source of the beta version is already released. See https://github.com/Ultimaker/Uranium and https://github.com/Ultimaker/Cura

MiguelAngelLV, 2015-05-13:
Thanks :+1:

BagelOrb, 2015-05-13:
setting name = xy_offset

MiguelAngelLV, 2015-05-13:
I try use the beta in Linux and it don't work the UI, the opengl crash.
CuraEngine by commands yes works. Can I  activate the xy_offset in the
15.02.1 version?

BagelOrb, 2015-05-13:
No, the setting is only included in the new Cura.
Since the new Cura hasn't officially released yet I can imagine it's hard to get it working.
Perhaps you can submit a bug report for the OpenGL crash? : https://github.com/Ultimaker/Uranium/issues

derekhe, 2015-06-07:
Why not implement this in this version?

BagelOrb, 2015-06-08:
The current master and 15.06 branch are for the new Cura. XY compensation is already implemented in 'this version'.