daid/LegacyCura #1195: New Machine Settings

ECALDIR, 2015-03-16:
Hi,

As a startup company we are launching our 3D printers soon. But we would like to add our machine settings to the Cura. Sorry for my noob question but what is the correct way to do it?

daid, 2015-03-16:
Sending a pull request with the proper added files in resources/machine_profiles.

closed 2015-03-16 by daid

ECALDIR, 2015-03-16:
Hi David;

Thanks for your fast response.

We also would like to add heated bed model in the profile. Which is located
in the /resources/meshes file. Right now we have two different machines.
Can we make the machine selection process like printrbot's. When you select
printrbot on the menu you can select from their different machines.

We can provide the necessary modifications on the source code. But we are
quşte unexperienced about github. Sorry for the trouble.

Thanks for your time.