daid/LegacyCura #1197: Support material isn't generated for small parts.

Nandox7, 2015-03-24:
When trying to enable support material for smaller parts the software doesn't generate it.
I've tried several combinations in the settings but none seems to work.

Some examples:
[image]

[image]

[image]

STL files used in the screenshots:
http://www.thingiverse.com/thing:654471
http://www.thingiverse.com/thing:673859

Nandox7, 2015-03-24:
Actually is even worse, it is adding support in the last place I'd like to have it. :(

[image]

BagelOrb, 2015-03-25:
As for your last problem: select 'touching buildplate' instead of ' everywhere' in the support type.

As for the other problems: they have all been fixed in the (yet to come) new version of Cura.

closed 2015-03-25 by BagelOrb