daid/LegacyCura #1201: Provide better support for printing outside of Cura

Nopik, 2015-03-31:
I'm using Cura for slicing and Host Repetitier for the printing. Each time I load model to Cura I need to choose 'Save GCode' from menu and then locate the resulting file and manually open it. With minimal changes Cura could streamline the process a lot: provide keyboard shortcut for 'Save GCode' and add 'open' icon (to open with system-default program registered for *.gcode) next to 'open folder' icon in the notification about gcode file being saved.

daid, 2015-04-01:
You can also right click on the save button, which has the save gcode option.

Nopik, 2015-04-01:
Thanks for the tip. It doesn't cut it, though. I'm the keyboard user, it is annoying that most of the File options have keyboard shortcuts, but the one which I'm using the most does not have ;)

daid, 2015-04-01:
Well, if you add "\tCTRL+G"
To the line in:
https://github.com/daid/Cura/blob/SteamEngine/Cura/gui/mainWindow.py#L84

You can use CTRL+G to call up the save with the keyboard.

Nopik, 2015-04-01:
Exactly. Any chances for it ending up in the official code?

Nopik, 2015-04-01:
Added #1205